adafruit/Adafruit_CircuitPython_HID #4: Doesn't work on gemma m0

ladyada, 2017-07-02:
works on feather m0 express but not gemma m0

Adafruit CircuitPython 1.0.0-rc.3-dirty on 2017-07-01; Adafruit Gemma M0 with samd21e18

```
from adafruit_hid.keyboard import Keyboard
from adafruit_hid.keycode import Keycode
import time

kbd = Keyboard()
kbd.press(Keycode.A)
time.sleep(0.1)
kbd.release_all()
```
no errors - just no keypress

tannewt, 2017-07-03:
This works for me except if I have the terminal up the first a is consumed when opening the repl. If I change it to type 5 As then I get:

```
Adafruit CircuitPython 1.0.0-rc.3-1-g3cd710c8-dirty on 2017-07-03; Adafruit Gemma M0 with samd21e18
>>>
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:

Adafruit CircuitPython 1.0.0-rc.3-1-g3cd710c8-dirty on 2017-07-03; Adafruit Gemma M0 with samd21e18
>>> aaaa
```

My file is:
```python
from adafruit_hid.keyboard import Keyboard
from adafruit_hid.keycode import Keycode
import time

kbd = Keyboard()

for i in range(5):
    kbd.press(Keycode.A)
    time.sleep(0.1)
    kbd.release_all()
```

dhalbert, 2017-07-03:
What happens if you just run that code manually from the REPL instead of as auto-reload?  All the testing I have done is from the REPL. 

If you insert a short sleep() before the loop, does the first `a` appear?

I haven't added sleep() before release_all(). You should be able to omit it.

tannewt, 2017-07-03:
Yes, all five `a`s are in the REPL when the same code is in REPL without the sleep.

dhalbert, 2017-07-03:
How about with the sleep in the REPL?  I am a little confused about whether it's the removal of the sleep or using the REPL that caused the problem to hp away. If the sleep is causing the problem maybe it's messing up the USB output in some way.

tannewt, 2017-07-03:
I don't think there is an issue at all. The first `a` is consumed by the "Press any key to enter the REPL".

dhalbert, 2017-07-03:
Ah! (slaps own forehead) (replying while on vacation one phone maybe not such a good idea)

ladyada, 2017-07-03:
hiya just to check - scott are you using the old gemma or are you using metro/feather?

tannewt, 2017-07-03:
The old gemma.

ladyada, 2017-07-03:
oh weird...ok maybe its me? ill do more testing tonite

tannewt, 2017-07-03:
Ok, keep me posted!

closed 2017-08-15 by ladyada