adafruit/Adafruit_CircuitPython_HID #5: Add multimedia keypress support

ladyada, 2017-08-15:
requires a different kind of report?
http://www.microchip.com/forums/m859944.aspx

dhalbert, 2017-08-15:
We need to add the "Consumer Usage Page" (0x0C) to the USB HID descriptor for the keyboard, and then the report is different as well. So there are changes both in CPy itself and in the library.

ladyada, 2017-08-15:
yah its a bit more work but MMKeys are really handy to make controllers for computers, i made a trinket + rotary encoder HID controller...would be rad if it controlled computer volume :)

dhalbert, 2017-08-15:
I found that project and code. Will look into this. ASF3 doesn't appear to support this out of the box - we'll have to augment it. I'll look at ASF4 too.

tannewt, 2017-08-15:
@dhalbert can you file an issue in adafruit/circuitpython for it and mark it for 3.0?

PR #12 (2018-04-03): Add Keyboard.send(); add ConsumerControl classes.
1. Add the convenience method `Keyboard.send()` which does `Keyboard.press()` and then `Keyboard.release_all()`. So:
```
kbd.send(Keycode.SHIFT, Keycode.A)
```
is the same as
```
kbd.press(Keycode.SHIFT, Keycode.A)
kbd.release_all()
```
This should be a lot more convenient to use.

2. Add `ConsumerControl` and `ConsumerControlCode` to send multimedia keypresses. New in CircuitPython 3.0. Tested on Linux.

3. Fixes two indentation issues that unnecessarily sent multiple reports for combination keypresses.

Fixes #5.
closed 2018-04-03 by tannewt