adafruit/Adafruit_CircuitPython_HID #6: missing 'SPACE' keycode

ladyada, 2017-08-15:
i think its '44' :)

dhalbert, 2017-08-15:
There's a `SPACEBAR` keycode, but I can add an alias for it named `SPACE` -- it's certainly more obvious! I was going by the names in http://www.usb.org/developers/hidpage/Hut1_12v2.pdf#page=58.

ladyada, 2017-08-15:
ooh! it doesnt appear in http://circuitpython.readthedocs.io/projects/hid/en/latest/keycode.html ?

dhalbert, 2017-08-15:
Oy, that's a doc issue. I didn't realize constants with no comments don't show up in readthedocs.

```BACKSPACE = 0x2A
    """Delete backward (Backspace)"""
    TAB = 0x2B
    """Tab and Backtab"""
    SPACEBAR = 0x2C
    MINUS = 0x2D
    """``-` and ``_``"""
```
`A` through `Z` are not in the docs either. I didn't add comments to them because I thought they were obvious.

I should check whether there's a setting to include definitions with no doc string.

tannewt, 2018-01-24:
Done in 6cdeb91a6b4b5a3ef828c5f64077c34bd4b57695

closed 2018-01-24 by tannewt