adafruit/Adafruit_CircuitPython_HID #10: Greater mouse movement

TheKitty, 2018-02-12:
Currently the mouse movement is limited to -127 to +127. Many screens are thousands of pixels long. Is this an artificial limit?

deshipu, 2018-02-12:
That's how far it can move in a single step. To move further, you have to do it in multiple steps, moving at most 127 units at a time.

TheKitty, 2018-02-13:
Limitation it is, unless there is thought of detecting a value greater than 127 and automatically handling it with multiple calls internally, it would be up to the user, case closed.

closed 2018-02-13 by TheKitty

tannewt, 2018-02-13:
@TheKitty Another option would be adding support for a touchscreen or something that does absolute positioning. We can't do dynamic descriptors yet but it'd be good to keep in the back of our minds.

ladyada, 2018-02-13:
absolute is a good one to add and i think its 'free' with the mouse HID descriptor we have. ill add an issue