adafruit/Adafruit_CircuitPython_HID #25: RawHID support

PTS93, 2018-07-26:
Something that would be very helpful in communicating with computer side frameworks as an alternative to VCP would be the possiblity to have a RawHID class.
My understanding of USB is not too in-depth but afaik this would just mean a new descriptor that allows for a report of the maximum possible size to send buffers back and forth.

Right now I can already do this with the existing classes, I'm just limited to very few bytes per transaction and can't receive something back.
F.e. with the consumer key device:
``` python
import usb_hid

value = 4024
code = bytearray(2)
bytes = value.to_bytes(2,'big')
for device in usb_hid.devices:
            if device.usage_page == 0x0C and device.usage == 0x01:
                print('found consumer device')
                myDevice = device
                break
device.send_report(bytes)
```

I found [this](https://docs.micropython.org/en/latest/pyboard/library/pyb.USB_HID.html) in the mpy docs but apart from this lonely docs page not much more.

tannewt, 2018-07-30:
I'd rather support a second cdc connection as tracked by: https://github.com/adafruit/circuitpython/issues/231

Please reopen if you need HID specifically.

PTS93, 2018-07-30:
Seems I can't reopen by myself but yes, HID specifically would be what I'm interested in. I want to avoid serial all together as it poses a lot of trouble in the use cases I have. HID has proven to be a very robust way of communication. I think the second CDC port wants to tackle a different issue.

tannewt, 2018-07-30:
Could you elaborate more on what you are doing then? Thanks!

PTS93, 2018-07-31:
Are OUT endpoints represented in the API yet? That seems like it would be the biggest chunk of work if that doesn't exist yet. Getting the descriptor going shouldn't be too hard, having a look at some Arduino implementation probably helps.
Teensy: https://github.com/PaulStoffregen/cores/blob/master/usb_rawhid/usb.c#L58-L90
AVR: https://github.com/NicoHood/HID/blob/master/src/SingleReport/RawHID.cpp#L26-L48

fgervais, 2019-02-20:
I'd also be interested in that feature

luolou, 2019-12-18:
Can Adafruit_CircuitPython_HID operate in Python? 
An error appears when I run this in Python:
**ModuleNotFoundError: No module named 'usb_hid'**
Anyone knows the reason?

PTS93, 2019-12-18:
@luolou thats a totally different issue but to answer that quickly, no if you are trying to run this on an SBC/Desktop, this will not work

PTS93, 2019-12-18:
@tannewt could this get another shot? @hathach added raw hid support in TinyUSB a while back :)

dhalbert, 2019-12-18:
I started working on raw HID for a special project but it's not at the top of my stack right now.

tannewt, 2019-12-18:
@deshipu started adding OUT support here: https://github.com/adafruit/circuitpython/pull/2366

It is very basic for numlock only.

luolou, 2019-12-18:
> @luolou thats a totally different issue but to answer that quickly, no if you are trying to run this on an SBC/Desktop, this will not work

@PTS93 I run this on laptop(windows10 OS). Does this only run on adafruit devices? Whether regular computers also need support?

PTS93, 2019-12-18:
@luolou this will only work on microcontrollers that have native USB device support and are supported by CircuitPython. No Laptop or Desktop computer will ever work with this library as they are not USB devices but USB hosts. This library only supports "emulating" HID devices which talk to a USB host like a laptop or desktop PC. E.g. a keyboard that plugs into your laptop. Your laptop is not capable of acting like a keyboard to another computer through a USB connection.
If you have further questions please post them in the Adafruit Forums: https://forums.adafruit.com/viewforum.php?f=60
Or on the Adafruit Discord.

luolou, 2019-12-18:
@PTS93 Thank you for your help.

dhalbert, 2021-05-04:
CircuitPython 7.0.0 will allow you to supply your own HID report descriptors, and so will support raw HID. See https://github.com/adafruit/circuitpython/pull/4689, which is still in process. That PR only provides one HID device descriptor, which means you can only have raw, and no other HID devices, but could be extended in the future to support multiple HID devices (one raw, one conventional with multiple devices). Please open an issue in circuitpython if you will need multiple HID devices. I will close this for now.

closed 2021-05-04 by dhalbert

timonsku, 2021-05-06:
@dhalbert cool, will this also take care of being able to listen to both in and out endpoints? Afaik there was no API for the in endpoint.

dhalbert, 2021-05-06:
There is now a [`.last_received_report` attribute](https://circuitpython.readthedocs.io/en/latest/shared-bindings/usb_hid/index.html#usb_hid.Device.last_received_report) which was added to read LED status on keyboards, but I think it would work for this

dhalbert, 2023-03-28:
https://github.com/adafruit/circuitpython/pull/7806 fixes a problem that gets in the way of raw HID, and should now allow this. There are some test programs there.