adafruit/Adafruit_CircuitPython_HID #32: Deprecated link to usb.org in adafruit_hid documentation

jpecor, 2019-03-26:
There's a link to usb.org in the adafruit_hid.keycode.Keycode section of the documentation:

http://www.usb.org/developers/hidpage/Hut1_12v2.pdf#page=58

This link appears to be deprecated and leads to a Page Not Found error.   I did some traversing around usb.org and found the **_Hut1_12v2.pdf_** document here:

https://www.usb.org/sites/default/files/documents/hut1_12v2.pdf

But I don't know for sure if that is the intended reference.

jpecor, 2019-03-26:
@tannewt  - Yes, I'm happy to create a PR.  Though, I don't know where to find the source for the readthedocs.io document.

dhalbert, 2019-03-27:
The RTD doc is generated from inline documentation in the code. The obsolete link is here: https://github.com/adafruit/Adafruit_CircuitPython_HID/blob/master/adafruit_hid/keycode.py#L35. Thanks!

tannewt, 2019-03-27:
Fixed by #33. Thanks @jpecor

closed 2019-03-27 by tannewt