adafruit/Adafruit_CircuitPython_HID #34: support for 16bit values in the gamepad module

marclava, 2019-04-28:
My jstest command on Linux shows 16bits values for signed 8bit values sent from the gamepad.py module; would it be possible to send 16bit values? I need a better resolution for my project...

dhalbert, 2019-04-28:
We'd like to generalize this so you can specify gamepads with wider ranges, more buttons, etc. This requires run-time generation of USB HID descriptors, which is a long-term goal of ours. See also:

https://github.com/adafruit/circuitpython/issues/1696
https://github.com/adafruit/circuitpython/issues/1015

marclava, 2019-04-28:
Being new to the wonderful world of HID devices, I just found that "Virtual Reality controls" are listed in the "USB HID Usage Tables", which corresponds more to what I'd like to create : https://www.usb.org/sites/default/files/documents/hut1_12v2.pdf , so a better question would be: how to create a (USB and BLE) HID VR device?