adafruit/Adafruit_CircuitPython_HID #48: Does not load on Gemma M0?

larsks, 2020-03-30:
I have a Gemma M0. I was attemping to follow through <https://learn.adafruit.com/adafruit-gemma-m0/circuitpython-hid-keyboard-and-mouse>, which claims to , but trying to `from adafruit_hid.keyboard import Keyboard` results in a memory allocation failure:

```
Adafruit CircuitPython 5.0.0 on 2020-03-02; Adafruit Gemma M0 with samd21e18
>>> from adafruit_hid.keyboard import Keyboard
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/lib/adafruit_hid/Keyboard.py", line 34, in <module>
MemoryError: memory allocation failed, allocating 832 bytes
>>> 
```

This is after a fresh boot, with nothing else loaded.

Is the `adafruit_hid` module expected to load on a Gemma M0? The article certainly seems to think so, but I'm not sure how to work around this error.

dhalbert, 2020-03-31:
Did you use the .mpy library from the bundle or the .zip file? The straight .py file may not be able to load.

larsks, 2020-04-01:
That was sort of my question :).  I was pulling it straight from the repository, so I only had the `.py` files handy.  After building the `.mpy` files with `mpy-cross` and installing those, things seem to work as expected. Sorry for the noise!

closed 2020-04-01 by larsks

dhalbert, 2020-04-01:
Glad you got it going!

The .mpy's are availabe in the release directory: https://github.com/adafruit/Adafruit_CircuitPython_HID/releases/tag/4.1.0, e.g. https://github.com/adafruit/Adafruit_CircuitPython_HID/releases/download/4.1.0/adafruit-circuitpython-hid-5.x-mpy-4.1.0.zip, so you don't need to mpy-cross them yourself. Also they are in the bundle: https://circuitpython.org/libraries.