adafruit/Adafruit_CircuitPython_HID #53: Refactor KeyboardLayoutUS to allow other layouts without duplicating code

dhalbert, 2020-08-19:
Factor out the common part of `KeyboardLayoutUS` into a common superclass, so that someone could add another keyboard layout just by specifying the ASCII (or Unicode?) to keycode table, and sharing the rest of the code. See #46 for the initial motivation for this.

Nlight91, 2021-03-01:
It would be loads of work. to simulate directly from hardware a non-us board. There are many layouts possible, french be is not the same as french fr, french ca is not the same as the two previous.

In this package, keycodes should be named scancodes, as it is these that it sends to the os. Scancodes are supposed to reflect the physical location of a key on the board. By convention, the us layout is set as reference. It is the os according to its own selected layout that translate the scancode into characters.

Dead keys are only possible at the os level, on azerty fr, the "ê" character is done by pressing first the dead key "^" located on the right of "P", and then pressing "E". If on the os the us layout is selected, and your board try to emulate azerty fr, to write "ê" it will press "[" then "E", the result will be very different.

Your suggestion is nice, but intimately tied the to the os layout, and from the board you have no way of knowing which is used.

What you suggest must have a goal in mind, what is it ? maybe there are other ways.

I believe TMK and/or QMK have localized aliases for scancodes, like FR_CIRCUMFLEX = LEFT_BRACKET. While that too is a lota work, it remains a more universal and conventional approach.

AngainorDev, 2021-03-18:
First attempt here https://github.com/adafruit/Adafruit_CircuitPython_HID/pull/61
Comments are welcome.