adafruit/Adafruit_CircuitPython_HID #58: Mouse not recognized on iPad

srcnet2, 2021-02-16:
I created a test sketch that works with Windows, Linux and Mac. When plugging into an iPad nothing happens. No cursor is shown on the screen.

Here is the sample code for testing:

`import time
import usb_hid 
from adafruit_hid.mouse import Mouse
import board
import digitalio

time.sleep(.5)
mouse = Mouse(usb_hid.devices)
time.sleep(.5)

led = digitalio.DigitalInOut(board.GP25)
led.direction = digitalio.Direction.OUTPUT

count = 0

while True:
    time.sleep(5)

    led.value = False
    time.sleep(.01)
    led.value = True
    while count<200:
        mouse.move(y=-5)
        mouse.move(x=-5)
        time.sleep(.01)
        count = count + 1
    count = 0
    while count<200:
        mouse.move(y=5)
        mouse.move(x=-5)
        time.sleep(.01)
        count = count + 1
    count = 0
    while count<200:
        mouse.move(y=5)
        mouse.move(x=5)
        time.sleep(.01)
        count = count + 1
    count = 0
    while count<200:
        mouse.move(y=-5)
        mouse.move(x=5)
        time.sleep(.01)
        count = count + 1
    count = 0`

Board / Firmware:
Adafruit CircuitPython 6.2.0-beta.2 on 2021-02-11; Raspberry Pi Pico with rp2040

dhalbert, 2021-02-16:
If I remember correctly, iOS does not like the gamepad support in the multi-device HID descriptor that CircuitPython presents. If you turn off `Gamepad`, I think it will work. But you'd need to make a custom build. See https://learn.adafruit.com/building-circuitpython/customizing-usb-devices and the rest of that guide.

js37, 2021-02-16:
Thank you so much. I was able to build CircuitPython and added:

`
USB_HID_DEVICES = "MOUSE" 
`

to the bottom of `mpconfigboard.mk` and it worked perfectly.

srcnet2, 2021-02-16:
Thank you so much. I was able to build CircuitPython and added:

`
USB_HID_DEVICES = "MOUSE" 
`

to the bottom of `mpconfigboard.mk` and it worked perfectly.

closed 2021-02-16 by tannewt