adafruit/Adafruit_CircuitPython_HID #59: Mouse input does not work (is ignored) in MacOS

marcos-diaz, 2021-03-07:
## Setup
* Firmware: CircuitPython 6.2.0 beta 3 (en_US)
* Board: Raspberry Pi Pico
* Host: Apple MacOS Big Sur 11.2.2 on MacBook Pro 2014

## Reproduce
1. Load this code as `code.py`
```
import time
import board
from digitalio import DigitalInOut, Pull
import usb_hid
from adafruit_hid.mouse import Mouse

button = DigitalInOut(board.GP16)
button.pull = Pull.DOWN
mouse = Mouse(usb_hid.devices)

while True:
    time.sleep(0.1)
    if button.value:
        print('CLICK!')
        mouse.click(Mouse.LEFT_BUTTON)
```

2. Connect to host OS via USB cable.
3. Push the button.

## Expected results
* Mouse clicks are performed on pushing the button.
* It should work in any operative system.

## Actual results
* MacOS recognises there is a mouse (see below), but fails to perform a mouse click.
* It does not work either in MacOS recovery mode (no user config).
* Works as expected on Windows 10.
* Works as expected on GNU/Linux.
* Works as expected on PC BIOS.
* No errors nor warnings are shown in the stdout.

## Additional notes.
* Mouse movement and wheel does not work either, with same results regarding OS' (it works in everything but MacOS).
* Mouse `usb_hid.device` is present as expected with 0x01 page and 0x02 usage.
* MacOS actually detects there is a mouse connected since the mouse config menu goes from this
[image]
to this
[image]
when connected.

Thanks.

marcos-diaz, 2021-03-09:
I also replicated the issue in an older Mac with 10.15 Catalina.

TomasHubelbauer, 2021-03-15:
FWIW, I can reproduce exactly the same issue, the only difference being I am on 11.2.3. Presumably, it worked very recently, since this YouTube video came out 10 days ago and in it, the library usage works:
https://www.youtube.com/watch?v=MjCFJCfq8ko

dhalbert, 2021-03-16:
@marcos-diaz Are you able and/or willing to try a custom build of CircuitPython? If so, try turning off other HID devices, particularly the "Gamepad" device. We know that on iOS, if Gamepad is on, other devices such as Keyboard don't work.

If you can't do a custom build, I can try one.

marcos-diaz, 2021-03-16:
@TomasHubelbauer thanks to the video you linked I noticed that CircuitPython 6.2.0 beta 2 does not have the issue and mouse works as expected in MacOS. So the bug was introduced in beta 3.

@dhalbert 👆above info is possibly more relevant than me trying to compile the whole project, ping me later if you want me to try nightly builds for potential fixes.