adafruit/Adafruit_CircuitPython_HID #65: Cannot use Control-C to interrupt code.py after sending string

jerryneedell, 2021-04-25:
I am using the code below on a neopixel_trinkey_m0 to send a string when either of the trinkey cap touch pads are touched.

I have found that it works fine, but if I have the REPL open then use then trigger it to send more than once, then I am no longer able to interrupt the execution of code.py from the REPL with a Control-C. Also if I try to exit my "screen" session on Linux, I am unable to close the session.  The code keeps running fine.. If I move my focus to a different window, I can see the string being printed on each touch.  I have to press RESET to get out of code,py.

I'm not sure if this is an issue with the library or with usb_hid.

I also had this same issue with a PIRKEY_M0 using irremote to trigger sending the string.

```
from adafruit_hid.keyboard import Keyboard
from adafruit_hid.keycode import Keycode
from adafruit_hid.keyboard_layout_us import KeyboardLayoutUS
import neopixel
import board
import time
import usb_hid
import touchio 

TOKEN = "0123456789012345678901234567890123456789"

led = neopixel.NeoPixel(board.NEOPIXEL,4,brightness=.3)

# The keyboard object!
time.sleep(1)  # Sleep for a bit to avoid a race condition on some systems
keyboard = Keyboard(usb_hid.devices)
keyboard_layout = KeyboardLayoutUS(keyboard)  # We're in the US :)

pad1 = touchio.TouchIn(board.TOUCH1)
pad2 = touchio.TouchIn(board.TOUCH2)
print("Ready! - touch to send code ")

while True:
    led.fill(0x100000)    # set LETs dimm  RED while waiting
    if pad1.value or pad2.value:
        led.fill(0x001010) # Change LED color-- cyan is pretty
        keyboard_layout.write(TOKEN)
        time.sleep(1)

```

dhalbert, 2021-04-25:
If you remove either or both of the touch or NeoPixel code, does the program go away? We would want to narrow it down to just touch, just usb_hid, or just NeoPixel, or maybe only some combination of two of the three.

jerryneedell, 2021-04-25:
here is simpler test - without touch  or neopixel -- it waits for an enter to start so you can get into the REPL
you can control-C to exit before the second time the string is sent -- after that control-C does not work

```
from adafruit_hid.keyboard import Keyboard
from adafruit_hid.keycode import Keycode
from adafruit_hid.keyboard_layout_us import KeyboardLayoutUS
import board
import time
import usb_hid

TOKEN = "0123456789012345678901234567890123456789"

# The keyboard object!
time.sleep(1)  # Sleep for a bit to avoid a race condition on some systems
keyboard = Keyboard(usb_hid.devices)
keyboard_layout = KeyboardLayoutUS(keyboard)  # We're in the US :)
input("press ENTER to start")
while True:
    print("sending string")
    keyboard_layout.write(TOKEN)
    time.sleep(10)

```

dhalbert, 2021-04-25:
I believe this is a core CircuitPython problem. Moving to circuitpython