adafruit/Adafruit_CircuitPython_HID #68: CAPSLOCK LED wont work

Jens1985, 2021-05-16:
My Code:
`import time
 
import board
import digitalio
import usb_hid
from adafruit_hid.keyboard import Keyboard
from adafruit_hid.keyboard_layout_us import KeyboardLayoutUS
from adafruit_hid.keycode import Keycode

keyboard = Keyboard(usb_hid.devices)
keyboard_layout = KeyboardLayoutUS(keyboard)  # We're in the US :)

keyboard.led_on(0x02)`

The Ouput:
`Traceback (most recent call last):
  File "code.py", line 14, in <module>
AttributeError: 'Keyboard' object has no attribute 'led_on'`

I think Keyboard has an attribute for led_on.

dhalbert, 2021-05-16:
If you get that error, you're probably using an older version of the library. `led_on() `was added in 4.2.0. Update to the latest version.

Jens1985, 2021-05-16:
I am using adafruit-circuitpython-hid-6.x-mpy-5.0.0.zip on `adafruit-circuitpython-hid-6.x-mpy-5.0.0.zip
`

dhalbert, 2021-05-16:
Ah. The NeoPixel Trinkey has HID frozen in, but the version frozen in is older and does not have `led_on()`. This will be fixed in 6.3.0, but for now, you can put the latest `adafruit_hid` in `/` instead of `/lib`, and it will override using the frozen library.

```py
>>> import sys
>>> sys.path
['', '/', '.frozen', '/lib']
```
`.frozen` comes before `/lib`, but after `/`.

Jens1985, 2021-05-16:
Thank you. It work now :)

closed 2021-05-16 by dhalbert