adafruit/Adafruit_CircuitPython_HID #71: Fix gamepad examples

dhalbert, 2021-05-29:
`gamepad.py` is no longer in the base library, because the gamepad device was removed from CircuitPython 7.0.0.

- Rename `examples/gamepad.py` to `examples/hid_gamepad.py` or similar, because `gamepad.py` overlaps with the `gamepad` built-in module.
- Fix or remove the gamepad-based examples.
- Add documentation about how to add a gamepad device to CircuitPython 7.0.0, or refer to a guide.

dhalbert, 2021-08-20:
There's now a gamepad example in https://learn.adafruit.com/customizing-usb-devices-in-circuitpython/hid-devices#custom-hid-devices-3096614-9.

dhalbert, 2021-08-20:
I think we should rename `examples/gamepad.py` to `examples/hid_gamepad.py`, or just discard it. The class name could stay the same.

I don't know whether we should just remove the other gamepad examples, or change them to use `hid_gamepad`. It seems odd to just discard the code, but they are kind of ancillary if there's no more `gamepad.py` in the library.

FoamyGuy, 2021-08-20:
I'm in favor of keeping them instead of discarding even though `gamepad.py` is not a part of the library itself. I think it would be a 
shame to get rid of it entirely, and I think this is the library that it is most closely related to so it seems like a logical place for that folks looking for something like that would be checking. 

I'll make a PR with those changes.

PR #76 (2021-08-21): fix gamepad examples
resolves #71 

The changes in this PR: 
- rename `gamepad.py` to `hid_gamepad.py`
- change the usage imports in the other examples to match the new name
- add note and guide link about needing to setup the gamepad HID device
- fix the `adafruit_seesaw` import. During testing I noticed that it was not able to be initialized properly using the code that was there before.

I tested `hid_gamepad.py` and the two examples that make use of it on a Feather RP2040 with 7.0.0 alpha6
closed 2021-08-21 by dhalbert