adafruit/Adafruit_CircuitPython_HID #72: .send() and .write() behavior on Windows vs Pi OS

BirchPrat, 2021-06-14:
I ran into problems when using the Capacitive Touch HID on the Pi4. The problem boils down to the following: code utilizing the .send() and .write() functions (on e.g. the NeoTrinkey) work perfectly well on my Windows machine (they send and write keyboard commands as they are supposed to), but neither send nor write anything when connected to a Raspberry Pi 4 running the most recent Pi OS.

I already posted the issue on the Adafruit forum, where a more detailed discussion of the topic can be found: 
https://forums.adafruit.com/viewtopic.php?f=52&t=180124&p=877286#p877286

dhalbert, 2021-06-14:
I read the forum post. Just to be clear, you are running the example on the Trinkey both times, right? The problem is that on Windows, you see the output generated ("A" and "Hello, World!\n") in a text-receiving window, but when doing the same thing when plugged into a Pi, you don't see that those outputs?

What kind of window is the output going to on Windows? Is it a PowerShell or CMD window, a text editor? ,etc.

On the Pi, are you using the RPi desktop? If so, which window is the output going into? Did you click inside that window to move keyboard focus to that window? Make sure the mouse pointer is inside the window.

I will try to replicate this when I hear the exact details.

BirchPrat, 2021-06-15:
> I read the forum post. Just to be clear, you are running the example on the Trinkey both times, right? The problem is that on Windows, you see the output generated ("A" and "Hello, World!\n") in a [...]

Exactly right, that was the behavior I experienced.

I just found out why I did not see any output on the RPi desktop . I am running the RPis headless and connecting with them via ssh (where I was obviously not expecting an output) and "Remote Desktop Connection" where I thought the output should be observable. As it turns out, the output does get transferred, but was simply not observable through Remote Desktop Connection, After hooking up a monitor to my raspberry pi I can observe the output as expected. My bad, I did not figure out right away that the connection via RDC to the Pi kept me from observing the output.

Thanks for you help, this issue can be closed now.

closed 2021-06-15 by FoamyGuy

dhalbert, 2021-06-15:
> the connection via RDC to the Pi kept me from observing the output.

Whether the local keyboard and mouse work depends on the technology of the remote desktop connection, and might even be configurable depending the the remote. In Windows RDP uses "Terminal Services" which creates a virtual connection for the remote keyboard, mouse, and display. On a Windows machine there can be a remote session happening simultaneously with a local session (licensing permitting).

For instance, VNC works differently, mirroring the actual local devices. It is connecting at a different level. See https://superuser.com/questions/32495/whats-the-difference-between-rdp-vs-vnc, for further info.