adafruit/Adafruit_CircuitPython_HID #78: Swedish Layout

XoidPLays, 2021-08-22:
Hi im trying to get this Payload working but i cant get this symbol (\) on the Swedish keyboard To work and i can't do this string (STRING %USERPROFILE%\hotdog.jpg)
This is what i have come up with rn

eng   swe
>   =  :
&   =  /
'   =  \
_   =  ?
)   =  =

Neradoc, 2021-08-22:
It looks like `\` is `altgr`-`+`  on a (windows) swedish keyboard, and `+` matches the position of the `-` key on a US, so you can send it with `keyboard.send(Keycode.RIGHT_ALT, Keycode.MINUS)` if I got that right. You would need to create a swedish `keyboard_layout_swe.py` version of the layout class to be able to `write()` it directly.

Note, this might be more a support question, they are better asked on [discord](https://adafru.it/discord) if you can (#help-with-circuitpython) or the [adafruit forums](https://forums.adafruit.com/).

XoidPLays, 2021-08-23:
Thx For Your Answer @Neradoc   Im not that good with python and shit like that and do u have a discord i can dm u on? 
github is a bit confusing sometimes
Here is my discord
(𝓧𝓸𝓲𝓭 𝓟𝓵𝓪𝔂𝓼 𝓨𝓣#9884)

Neradoc, 2021-08-23:
Come to the Adafruit discord server, many people can help there.
https://adafru.it/discord

XoidPLays, 2021-08-23:
i joined i just dont know what to start with, im new to this

dhalbert, 2021-08-23:
Followed up in discord, so closing here.

closed 2021-08-23 by dhalbert