adafruit/Adafruit_CircuitPython_HID #90: Add shuffle mode

qik33, 2021-12-29:
Would it be possible to add shuffle to the available options?

I assume it's the RANDOM_PLAY function from the HID Usages and Descriptions document, so it would probably look like this (?):

`RANDOM_PLAY = 0xB9
    """Shuffle"""`

Thanks!

dhalbert, 2021-12-29:
We've avoid adding more constants to the library because each name makes the library larger, and there are hundreds of names. We are very tight on space on some of the smallest builds which still want to provide the library. You could just use `0xB9` as a constant, of course, it doesn't have to be named. Did it work for you?

This comment mentions where to look for more codes:
https://github.com/adafruit/Adafruit_CircuitPython_HID/blob/c36ae716de1ad587142e3b09ef09206fffb2ab6d/adafruit_hid/consumer_control_code.py#L17

qik33, 2021-12-29:
Thank you, I got it to work by using the unnamed constant!

closed 2021-12-29 by qik33