adafruit/Adafruit_CircuitPython_HID #92: Mac Fn Key Support

edm00se, 2022-01-24:
I'd like to be able to emit the media control key press, such as pause/play Fn + F7. From my looking through `adafruit_hid.keycode` and documentation I don't currently believe this is supported but I'd love to see it made available.

Neradoc, 2022-01-24:
Media keys are Consumer Controls. There is a short list available in the library:
https://docs.circuitpython.org/projects/hid/en/latest/api.html#adafruit-hid-consumer-control-code-consumercontrolcode
And you can supply any valid value as ints.

Here is the example code for ConsumerControl:
https://github.com/adafruit/Adafruit_CircuitPython_HID/blob/main/examples/hid_consumer_control_brightness.py

edm00se, 2022-01-24:
Perfect! I'll need to re-work my macro but I think that example should get me headed in the right direction. Thank you.

closed 2022-01-24 by edm00se