adafruit/Adafruit_CircuitPython_HID #93: Not connecting USB to host causes either hang or crash

CarbonNeuron, 2022-02-02:
I am writing a script that needs to run when the computer first turns on. When it is still turning on sometimes there is a large delay between when power is given to the board and when the USB controller in the computer wakes up. I am using the keyboard module and mouse module. When I was running circuitpython on my neotrinkey, and the computer was taking longer then normal to turn on, it would start flashing some code. I do not have a second pico to debug on. 
This is a pretty niche use case but it also will happen if you just connect your pico to power and no data. 
Maybe there is a way to tell when the host is ready to accept connection or throw an error if it is not ready?
That way you can just loop the connect command or return a message if there is no host device.

CarbonNeuron, 2022-02-02:
As reference this will happen if you run https://github.com/adafruit/Adafruit_CircuitPython_HID/blob/2fddabcaf0df1763111ed9dbf9e2d4cdb5b0434e/examples/hid_keyboard_shortcuts.py when you only have power hooked up to the board and no data.

CarbonNeuron, 2022-02-03:
Probably has something to do with #13

dhalbert, 2022-02-03:
Could you paste your simplest working example? Thanks.

CarbonNeuron, 2022-02-06:
I ended up just forking and fixing for my use case. Sorry about that.

closed 2022-02-06 by CarbonNeuron

ladyada, 2022-02-06:
@CarbonNeuron if its a bug, please submit a PR for the fix, it would be helpful :)

CarbonNeuron, 2022-02-06:
> @CarbonNeuron if its a bug, please submit a PR for the fix, it would be helpful :)

It's not really a bug. My use case of powering a board without the data lines of USB connected is very specific. 
And it can be handled normally with try catch.