adafruit/Adafruit_CircuitPython_HID #95: `find_device` module missing in `adafruit_hid.keyboard.Keyboard`

ivogeorg, 2022-02-08:
https://github.com/adafruit/Adafruit_CircuitPython_HID/blob/2fddabcaf0df1763111ed9dbf9e2d4cdb5b0434e/adafruit_hid/keyboard.py#L17

Doesn't the line `from . import find_device` in `adafruit_hid.keyboard.Keyboard.py` expect that there is a `find_device` submodule in the `adafruit_hid` library hierarchy? Or am I missing something about importing Adafruit libraries. I am only trying to execute the `code.py` on an Adafruit Metro M4 Express board with updated bootloader and CircuitPython.

dhalbert, 2022-02-08:
`find_device()` is a function, not a module. It is in https://github.com/adafruit/Adafruit_CircuitPython_HID/blob/main/adafruit_hid/__init__.py

ivogeorg, 2022-02-08:
Right, I see it. This is the error message I get:
```
code.py output:
Traceback (most recent call last):
  File "code.py", line 15, in <module>
  File "adafruit_hid/keyboard.py", line 17, in <module>
ImportError: no module named 'adafruit_hid.find_device'
```

First 16 lines of  `code.py`:
```python
# Metro M4 IO demo
# Welcome to CircuitPython 3.1 :)

import time
import board
from digitalio import DigitalInOut, Direction, Pull
from analogio import AnalogIn
from adafruit_motor import servo
import neopixel
import audioio
import pulseio
import simpleio

# keyboard support
from adafruit_hid.keyboard import Keyboard
from adafruit_hid.keycode import Keycode
```

ivogeorg, 2022-02-08:
I am using the `mpy` files from the 7.*.* library bundle.

dhalbert, 2022-02-08:
Is `__init__.py` in `lib/adafruit_hid`, and is it non-empty?

ivogeorg, 2022-02-08:
@dhalbert, I think I know what happened. When I was updating the libraries, a stray _empty_ `__init__.py` had remained in the `adafruit_hid`. As I said, I am using the compressed `mpy` files, so I can't view them. Deleted the stray and that cleared the import problem.

I am guessing this code is a bit outdated because I am running into errors that can be explained easily with changes in the libraries. 

Thanks for the prompt responses!

closed 2022-02-08 by ivogeorg

dhalbert, 2022-02-09:
Glad you figured it out.!A `.py` file takes precedence over a `.mpy` file of the same name, so what happened makes sense.