adafruit/Adafruit_CircuitPython_HID #96: Double key in Discord shortcuts

Neradoc, 2022-03-23:
There is a weird effect that happens in Discord shortcuts when sending a key. (Seen on mac).
This a simple `kbd.send(Keycode.R)` code will cause the shortcut to be recorded as "R + R". Pressing the key with the keyboard gets a normal "R".
It might be an issue with discord, but the discrepancy between the keyboard and adafruit_hid makes me wonder what is different in CP to cause that. It might very well be a builtin usb_hid issue.
- adafruit_hid
[image]

- keyboard
[image]

dhalbert, 2022-03-23:
If you do press/release, do you see that too? How about press/delay/release?

Neradoc, 2022-03-23:
Yes, no matter the delay between press and release.
Note that "R + R" seems to be meaning "simultaneously R and R", which is weird. Mashing 2 keys on the keyboard I manage to get "R + H + R + H" (same as `kbd.send(Keycode.R, Keycode.H)`), but I can't get "R + R" on the keyboard.

Overall it seems like a mac-discord only thing, but I'm really puzzled how that happens...