adafruit/Adafruit_CircuitPython_HID #99: Control the speed of layout.write()?

tobiashochguertel, 2022-08-03:
Is there a way to controll the speed of layout.write(), so that it runs faster? or is there a way to improve the speed of circuitpython on my rasperrypi pico?

I want to do a VS Code Keyboard, which runs things like:

```
while True:
    led.value = False
    if button.value:
        kbd.send(Keycode.COMMAND, Keycode.SHIFT, Keycode.P)
        # layout.write("Transform to Uppercase\n")
        layout.write("Uppercase\n")
        kbd.release_all()
        led.value = True
    time.sleep(0.2)
```

But it's reacting slow, or with a delay.

dhalbert, 2022-08-04:
Could you describe more precisely what is happening in terms of delays? Is that the whole program? Is the `time.sleep(0.2)` for debouncing?

Consider using the `keypad` module instead of manual debouncing, so you don't need delays: https://learn.adafruit.com/key-pad-matrix-scanning-in-circuitpython

The `kbd.release_all()` is not necessary, because both `.send() and `.write()` will release keys. `.write()` simply does a lookup for each character and then does a .`send()` or the equivalent.

tobiashochguertel, 2022-08-04:
the `time.sleep(0.2)` is for the LED, to switch on and off - to get a feedback about a button press.

I tried also:

```
        kbd.send(Keycode.COMMAND, Keycode.SHIFT, Keycode.P)
        # layout.write("Transform to Uppercase\n")
        # layout.write("Uppercase\n")
        kbd.send(Keycode.U)
        kbd.send(Keycode.p)
        kbd.send(Keycode.p)
        kbd.send(Keycode.e)
        kbd.send(Keycode.r)
        kbd.send(Keycode.c)
        kbd.send(Keycode.a)
        kbd.send(Keycode.s)
        kbd.send(Keycode.e)
        kbd.send(Keycode.ENTER)
```

because that is maybe faster - but, that doesn't work - only the first letter is pressed.

[image]

My issue with ".write" is that it has a delay until the VS Code Command "Trasnform to Uppercase" is done, it's like I press the button on my breadboard, and then I have to wait until my selection is transformed into uppcase. The thing is that, it is still super fast, because I can't see any interaction with VS Code - but the effect takes some time to be then fired.

I created for "Transform to Uppercase" a shortcut, and let my Raspberry Pi Pico press the shortcut - and that gets triggered / transformed direct without waiting.

```
kbd.send(Keycode.LEFT_CONTROL, Keycode.LEFT_ALT, Keycode.COMMAND, Keycode.KEYPAD_ONE)
```

closed 2023-09-06 by tannewt