adafruit/Adafruit_CircuitPython_HID #105: ImportError: no module named '__future__'

ElectricCoffee, 2022-10-17:
Hi, I'm getting the following error when trying to follow the MacroPad Basics tutorial:

```
code.py output:
Traceback (most recent call last):
  File "code.py", line 9, in <module>
  File "adafruit_macropad.py", line 62, in <module>
  File "adafruit_hid/__init__.py", line 21, in <module>
ImportError: no module named '__future__'
```

I've tried both the v7 and v8 files, and I get the same result both times.

I copied the files onto the board as instructed

dhalbert, 2022-10-17:
`__future__` was added to CircuitPython sometime in 7.x. I tested this with 7.3.0 and with 8.0.0-beta.2, and I see `__future__`.

To find out what version of CircuitPython is running, look in `boot_out.txt`. Make sure it corresponds to what you have loaded. If you see .uf2 files in CIRCUITPY, then you are not in bootloader mode when you are copying the files.

ElectricCoffee, 2022-10-17:
My version is 7.1.1

I actually went ahead and replaced `__future__.mpy` with `__future__.py` downloaded off of here and commented out the line that imported annotations, and it just works.

dhalbert, 2022-10-17:
If you upgrade to 7.3.3, you will have the builtin `__future__`.

ElectricCoffee, 2022-10-18:
You should update the docs to specify 7.3.x rather than 7.x.

7.x implies 7.1 will work, and that was the assumption I went by.

dhalbert, 2022-10-18:
I searched in the MaroPad guide, and there are a lot of general mentions of 7.x, but they come from guide pages that are shared and are not specific. I don't see one specific about the use of HID with MacroPad. Let us know any specific reference you found to be incorrect.

In general, the libraries track the latest stable version of CircuitPython, because there are bug fixes and improvements. If you get an error on something that should work, updating to the latest stable version is almost always a good idea. We can add that advice somewhere

ElectricCoffee, 2022-10-18:
All I can say is that I downloaded the project bundle on this page: [https://learn.adafruit.com/adafruit-macropad-rp2040/macropad-basics](https://learn.adafruit.com/adafruit-macropad-rp2040/macropad-basics), and upon installing the files found in the 7.x folder it came up with that error.

So even if it doesn't explicitly use the HID portions of the code, it seems to do so implicitly