adafruit/Adafruit_CircuitPython_HID #106: Control Codes in keyboard_layout_us not supported

bborncr, 2022-11-09:
Is there a reason that control codes besides LF (\n) are zeroed out in `keyboard_layout_us.py`?

For example, changing `b"\x00"  # CR \r` to `b"\xa8"  # CR \r` on line 48 allows \r to be used instead of \n

Now \r can be used to to send ascii art to a chat without breaking every line into separate messages:
```
ascii_art2 = \
"```           ___\r"\
"     |     | |\r"\
"    / \    | |\r"\
"   |--o|===|-|\r"\
"   |---|   | |\r"\
"  /     \  | |\r"\
" |       | | |\r"\
" |       |=| |\r"\
" |       | | |\r"\
" |_______| |_|\r"\
"  |@| |@|  | |\r"\
"___________|_|_\r```"\
```
The result:
[image]

dhalbert, 2022-11-09:
The `ASCII_TO_KEYCODE` table in `keyboard_layout_us.py` is meant to map to physical keys. I looked up keycode `\xa8`, and it's something obscure. Did you mean `\x28`?

Assuming you meant `\xa8`, what chat system are you using that uses `\xa8` as a non-breaking newline?

bborncr, 2022-11-09:
`\0xa8` is `SHIFT+LF` `(SHIFT+0x28)` which I always understood was `\r` (a carriage return without a `LF`). But looking it up the actual ASCII code for `\r` is `15` (`0x0f`). `ENTER` is being send for `\n`, and I'm just doing a `SHIFT+ENTER` for `\r`. So maybe what I'm doing is just a hack?

dhalbert, 2022-11-09:
`SHIFT+LF` is a convention for non-breaking newline, but it doesn't really have anything to do with `CR` (`\r`). You could send a line of the image and then `kbd.send(Keycode.SHIFT, Keycode.ENTER)`, then another line, etc So, yes, it is a kind of a hack :slightly_smiling_face: .

If I were coding this up, I might send a list or tuple of the lines of the art, without `\r` in them. Then `write()` each one, one at a time, followed by the `kbd.send()` I gave above.

bborncr, 2022-11-09:
Understood! Thanks!

closed 2022-11-09 by bborncr