adafruit/Adafruit_CircuitPython_HID #110: Consumer Control not working

AbdElhalim12, 2022-11-29:
I'm trying to use consumer controls with pushbuttons
but I'm not getting anything.

I've tried the pushbutton with macros and it worked.
```
import time
import usb_hid
from adafruit_hid.keycode import Keycode
from adafruit_hid.keyboard import Keyboard
import board
import digitalio
from adafruit_hid.consumer_control import ConsumerControl
from adafruit_hid.consumer_control_code import ConsumerControlCode

> btn1_pin = board.GP9

> btn1 = digitalio.DigitalInOut(btn1_pin)
btn1.direction = digitalio.Direction.INPUT
btn1.pull = digitalio.Pull.DOWN

while True:

    if btn1.value:
        keyboard.send[[ConsumerControlCode.PLAY_PAUSE]]
        time.sleep(0.1)
```

dhalbert, 2022-11-29:
You cannot send `ConsumerControl` codes via `Keyboard()`. You need to create a `ConsumerControl` object and use that.

See https://github.com/adafruit/Adafruit_CircuitPython_HID/blob/main/examples/hid_consumer_control_brightness.py for an example. There are also many examples in https://learn.adafruit.com.

closed 2022-11-29 by dhalbert

AbdElhalim12, 2022-11-29:
hey, thanks for the solution.
one more question, when to use this line then?
[image]

dhalbert, 2022-11-29:
The MacroPad library interprets the items in that list as various different things. It doesn't have anything to do with ConsumerControl per se. See https://learn.adafruit.com/macropad-hotkeys?view=all#media-control-consumer-control-codes-3104573. ConsumerControl codes and Keyboard codes are both just integers.