adafruit/Adafruit_CircuitPython_HID #120: Implement HID LampArray Support

AStellarScene, 2023-10-04:
The new Dynamic Lighting feature in Windows 11 was just released. It uses a new standard called HID LampArray. This open standard aims to improve the overall lighting ecosystem for PC peripheral devices. Implementing this new standard into the current HID library would make it much easier to experiment with this new feature. My personal assumption is that almost all CircuitPython HID projects implement some form of RGB LEDs. For example, the Adafruit MacroPad implements many RGB LEDs, and this feature would be particularly useful.

[Microsoft Support - Control your Dynamic Lighting devices in Windows](https://support.microsoft.com/en-us/windows/control-your-dynamic-lighting-devices-in-windows-8e8f22e3-e820-476c-8f9d-9ffc7b6ffcd2)

[Microsoft Learn - Dynamic Lighting Devices](https://learn.microsoft.com/en-us/windows-hardware/design/component-guidelines/dynamic-lighting-devices)

[USB.org - HID Usage Tables for Universal Serial Bus (USB) Version 1.4](https://www.usb.org/sites/default/files/hut1_4.pdf)

henrygab, 2024-03-23:
This would be AWESOME!  Looks like there's a [C++ sample](https://github.com/microsoft/RP2040MacropadHidSample) using the Adafruit MacroPad.  Sadly, it's not trivial to use yet, at least due to a delayed [tinyusb PR](https://github.com/hathach/tinyusb/pull/2252).

If this gets enabled via CircuitPython ... Wow!

dhalbert, 2024-03-23:
I don't think we need the tinyusb changes: this looks like standard HID. It only requires a CircuitPython library to be written.

Cellgalvano, 2025-10-23:
Hi!

I would highly appreciate if this would be possible. For PC modding and custom led lights for PCs this would be a game changer to be able use circuitpython.

Is it possible to do this in python using the current library? 

BR