adafruit/Adafruit_CircuitPython_HID #121: BLE Keyboard Functionality Issue

Willie-Corto, 2023-11-29:
### CircuitPython version

```python
Adafruit CircuitPython 8.2.7 and 9.0.0-alpha4; ItsyBitsy NRF52840 Express
```

### Code/REPL

```python
#https://learn.adafruit.com/ble-hid-keyboard-buttons-with-circuitpython/ble-keyboard-buttons-libraries-and-code

import time
import board
from digitalio import DigitalInOut, Direction, Pull
import adafruit_ble
from adafruit_ble.advertising import Advertisement
from adafruit_ble.advertising.standard import ProvideServicesAdvertisement
from adafruit_ble.services.standard.hid import HIDService
from adafruit_ble.services.standard.device_info import DeviceInfoService
from adafruit_hid.keyboard import Keyboard
from adafruit_hid.keyboard_layout_us import KeyboardLayoutUS
from adafruit_hid.keycode import Keycode

hid = HIDService()
device_info = DeviceInfoService(software_revision=adafruit_ble.__version__,
                                manufacturer="Adafruit Industries")
advertisement = ProvideServicesAdvertisement(hid)
advertisement.appearance = 961
scan_response = Advertisement()
scan_response.complete_name = "CircuitPython HID"

ble = adafruit_ble.BLERadio()
k = Keyboard(hid.devices)

# Start Advertising
if not ble.connected:
    ble.start_advertising(advertisement, scan_response)

# Main Loop
while True:    
    k.send(Keycode.Q)        
    time.sleep(0.5)
```

### Behavior

When the ItsyBitsy NRF52840 Express is connected via USB to a PC, BLE keyboard functionality works as expected, allowing the device to send keystrokes to connected devices. However, when powered by a TP4056 module, a 3.7V lipo battery pack, or a USB charger (without data connection), the board will advertise BLE services and can be successfully paired, but it will not send keystrokes. 

### Description

Encountering an issue with the ItsyBitsy NRF52840 Express where BLE keyboard commands are not transmitted unless the device is connected to a PC via USB. Despite successful BLE service advertisement and pairing with external devices when powered through alternative means (such as a TP4056 module, a 3.7V lipo battery, or a USB charger without data), keystrokes are not sent. This problem persists across different boards and CircuitPython versions.

### Additional information

The problem was replicated on other boards: Seeed Studio XIAO nRF52840 Sense and Feather nRF52840 Express.
Bootloaders 0.7.0 and 0.8.0 were tested on all devices.
The issue was confirmed by another Adafruit forum user, who replicated it on their ItsyBitsy NRF52840.
https://forums.adafruit.com/viewtopic.php?t=205776
Original project: https://learn.adafruit.com/ble-hid-keyboard-buttons-with-circuitpython/ble-keyboard-buttons-libraries-and-code

dhalbert, 2023-11-29:
There is a fix in #122, not yet reviewed and released.

closed 2023-11-29 by dhalbert

dhalbert, 2023-11-29:
This is now released in https://github.com/adafruit/Adafruit_CircuitPython_HID/releases/tag/6.0.2 and will be updated in the bundle tonight.

PR #122 (2023-11-29): Do not wait for USB for non-USB HID devices
Fixes #121.

`find_device()` was waiting for USB to be connected even if the device was not a USB device. Wait only if a USB device was found.

In the long run, maybe the dependency on USB should be decoupled from `find_device()`. In that case, there should be some other way, maybe explicit or implicit, of waiting for the HID transport channel to be ready.

Also there could be an addition to `circuitpython_typing` for duck-typing an HID device.

I tested this on a CLUE.