adafruit/Adafruit_CircuitPython_HID #127: Modifiers not working across keyboards

alana314, 2024-08-13:
Hiya,
I'm making a tiny macro keyboard with modifier keys using CircuitPython on a Seeed Studio Xiao. It works, but when I hold down the modifier buttons and press a letter on a second keyboard, the keyboard buttons aren't modified. For instance, if I hold down button 5 (shift) and hit 'a' on my mac keyboard, I don't get a capital A.  
My relevant code looks like this:
```
    if btn5.value == False and btn5_state == False:
        print("Shift")
        btn5_state = True
        kbd.press(Keycode.SHIFT)
    elif btn5.value == True and btn5_state == True:
        print("Release Shift")
        btn5_state = False
        kbd.release(Keycode.SHIFT)
```

My full code is here: https://github.com/alana314/thumbKeys/blob/main/main.py  
When I test two real keyboards, holding down shift on one modifies the other. Is there something with my code, or is there something in the library that's releasing the button early?
Thanks

tannewt, 2024-08-13:
@dhalbert May know about this. I don't know how multiple keyboards are supposed to interact. Your code and the library look like they should do what you expect.

dhalbert, 2024-08-14:
I tried this on a Mac Mini M1 running Sonoma, with two standard PC keyboards plugged in. Pressing down the shift key on one keyboard did _not_ enable shift on the other keyboard, but only on the keyboard that with shift being pressed.

In your own tests, is one of the keyboards a Macbook laptop keyboard? That keyboard is not probably not connected internally via USB, but via some other method, and so the modifiers may work differently.. Or, if you are using Mac USB or Bluetooth keyboards, they may be being treated specially.

alana314, 2024-08-14:
Thanks for testing!  
So I tried on Windows, which is my intended use case, and it DOES work. So it must be functionality of the OS.  
Thanks!

closed 2024-08-14 by alana314