adafruit/Adafruit_CircuitPython_HID #129: Support absolute mouse coordinates (`adafruit_hid.mouse_absolute`)

bluepuma77, 2024-10-17:
We want to control mouse and keyboard on tablets connected via USB to an ESP32 S3 Box (or similar).

[adafruit_hid](https://docs.circuitpython.org/projects/hid/en/latest/) supports keyboard and relative mouse. As in [another issue](https://github.com/adafruit/circuitpython/issues/7176) mentioned, sometimes the need arrises to move to absolute positions to know where the mouse really is.

Work-arounds, like moving relative to (-5000;-5000) all the way top left to get to a fixed point, sometimes have issues, like iPads opening some additional layers and windows, sometimes they close again or stay open, therefore no real automation is possible.

It would be great to have `adafruit_hid.mouse` and additional a real `adafruit_hid.mouse_absolute` (probably as two separate USB devices) to be able to make relative and absolute moves with a mouse HID.

dhalbert, 2024-10-18:
Yes, the 8.x and 9.x libraries are compatible. Or you could use the .py version.

One reason this is not in `adafruit_hid` is because we need to keep that library small: it is frozen into some boards with limited flash and we don't have room for more features like absolute mouse or gamepad.

bluepuma77, 2024-10-18:
@dhalbert Would it be possible to create a regular and an absolute mouse at the same time?

The example tells to enable keyboard and mouse in `boot.py`:
```
import usb_hid
from absolute_mouse.descriptor import device

usb_hid.enable((usb_hid.Device.KEYBOARD, device))
```
Can I enable 3 HID devices: keyboard, regular mouse and absolute mouse? How?

dhalbert, 2024-10-18:
Yes, you can do 
```py
usb_hid.enable((usb_hid.Device.KEYBOARD, usb_hid.Device.MOUSE, device))
```
 **But**, you have to be a little trickier when passing a list of devices to `adafruit_hid.mouse.Mouse()` and `absolute_mouse.Mouse()`. Both mouse devices have the same Usage Page and Usage. They both have this in their report descriptors:
```
        0x05, 0x01,    # Usage Page (Generic Desktop)
        0x09, 0x02,  # Usage (Mouse)
```
Usage Page = 1, and Usage = 2

Normally you would do `Mouse(usb_hid.devices)`. But that looks for the _first_ device with the Usage Page and Uage values expected. So instead of passing the whole `usb_hid.devices`, just pass a tuple or list with the device you mean to use. If you are using the exact device list I gave above, the keyboard will be `usb_hid.devices[0]`, the relative mouse will be `usb_hid.devices[1]`, and the absolute mouse will be `usb_hid.devices[2]`. So do something like:

```py
absolute_mouse = absolute_mouse.Mouse((usb_hid.devices[2],))  # pass tuple just containing the absolute mouse device
```

bluepuma77, 2024-10-18:
When using [v8 files from release](https://github.com/Neradoc/CircuitPython_Absolute_Mouse/releases) and
```
from absolute_mouse.descriptor import device
```
I do get an error
```
incompatible .mpy file
```
Can I just copy `__init__.py` and `descriptor.py` into `/lib/absolute_mouse`?

dhalbert, 2024-10-18:
Yes, it's fine to use the .py versions.

bluepuma77, 2024-10-18:
Awesome, it works! I can move the mouse absolute on iPad. Just copied the two .py files into the `/lib/absolute_mouse` folder and added the initialization in `boot.py` from the example.

Thanks @dhalbert @jepler 

It would be great if the [HID doc](https://docs.circuitpython.org/projects/hid/en/latest/) could hint to this, even though it's not in the official package.

closed 2024-10-18 by bluepuma77

bluepuma77, 2024-10-18:
Had to apply the [patch](https://github.com/Neradoc/CircuitPython_Absolute_Mouse/issues/1) to make m.click() work.