adafruit/Adafruit_CircuitPython_HID #132: Unable to send ñ

marco-gallegos, 2024-12-26:
hello im using a pico with rp2040 and in not able to use this character.

i just read about on your docs and looks like is not working.

i tried using COMBINED_KEYS but is not fully working im on a us keyborad configuration and my current testing code is very simple just catching a 4x4 keyboard input and firing hid functions.

there exist any example or doc explaining this?
im doing this wrong?

'''python
import time
from digitalio import DigitalInOut
import board
import adafruit_matrixkeypad # v 1.2.17

import random
import digitalio
import usb_hid
from adafruit_hid.mouse import Mouse
from adafruit_hid.keyboard import Keyboard
from adafruit_hid.keyboard_layout_us import KeyboardLayoutUS as KeyboardLayout
# from keyboard_layout_win_es import KeyboardLayout
from adafruit_hid.keycode import Keycode
# from keycode_win_es import Keycode

## matrix values 
# Extended 4x4 matrix keypad
cols = [DigitalInOut(x) for x in (board.GP0, board.GP1, board.GP2, board.GP3)]
rows = [DigitalInOut(x) for x in (board.GP4, board.GP5, board.GP6, board.GP7)]
keys = ((1, 2, 3, "A"), (4, 5, 6, "B"), (7, 8, 9, "C"), ("*", 0, "#", "D"))

keypad = adafruit_matrixkeypad.Matrix_Keypad(rows, cols, keys)

## USB HID setup
led = digitalio.DigitalInOut(board.LED)
led.direction = digitalio.Direction.OUTPUT

# Inicializar los objetos Mouse y Keyboard, y KeyboardLayout
mouse = Mouse(usb_hid.devices)
keyboard = Keyboard(usb_hid.devices)
keyboard_layout = KeyboardLayout(keyboard)
keyboard_layout.COMBINED_KEYS = {
    "n": "ñ"
    # "ñ": "~+n",
    # "~+n": "ñ",
    # "en": "marco"
}

### functions hid section

def send_typing(text):
    keyboard_layout.write(text)

## main cicle
while True:
    keys = keypad.pressed_keys
    if keys:
        print("Pressed: ", keys)
    if "D" in keys:
        print("will send anie on usb")
        #send_typing("~+n")
        #keyboard.press(Keycode.POUND, Keycode.N)
        #keyboard.release_all()
        #print(keyboard_layout.keycodes("ñ"))
        #send_typing("ñ")
        send_typing("~n")
    time.sleep(0.1)
'''

Neradoc, 2024-12-26:
Hi, HID can only simulate key presses, and therefore type characters that can be typed in your PC's keyboard layout.
How do you type a `ñ` on your keyboard ? You can reproduce the same sequence.
As far as I know, if your keyboard is configured as windows US you would need something like Alt+164:

```py
keyboard.press(Keycode.ALT)
keyboard.send(Keycode.ONE)
keyboard.release(Keycode.ONE)
keyboard.send(Keycode.SIX)
keyboard.release(Keycode.SIX)
keyboard.send(Keycode.FOUR)
keyboard.release(Keycode.FOUR)
keyboard.release_all()
```

Borrowing ideas from the [Adafruit Hotkeys learn guide](https://learn.adafruit.com/macropad-hotkeys/project-code) for example you could code it something like this:

```py
macro = [Keycode.ALT, Keycode.ONE, -Keycode.ONE, Keycode.SIX, -Keycode.SIX, Keycode.FOUR]

def type_sequence(sequence):
    for key in sequence:
        if key>0:
            keyboard.press(key)
        else:
            keyboard.release(-key)
    keyboard.release_all()

type_sequence(macro)
```

This library doesn't include transforming characters into such sequences of keys.
There's a (as of yet undocumented) helper in the layouts repository.
https://github.com/Neradoc/Circuitpython_Keyboard_Layouts/blob/main/libraries/helpers/alt_codes.py

marco-gallegos, 2024-12-27:
```markdown
Hi, HID can only simulate key presses, and therefore type characters that can be typed in your PC's keyboard layout
```

this makes sense i just replied the sequence to write this character and worked, thank you a lot :)

as note i just triggered a keyboard shortcut to switch layout and then write the ';' key, this key contains the 'ñ' on the LAT layout, not so cool but works right now.

closed 2024-12-28 by marco-gallegos