adafruit/Adafruit_CircuitPython_HID #133: keyboard_layout.write issues on linux

mcglonelevi, 2025-02-27:
Hello,

I have some code that works as expected on Windows but does not send the keystrokes that I supply on Linux.  Here's my code.py:

```
import time
import board
import digitalio
import usb_hid
from adafruit_hid.keyboard import Keyboard
from adafruit_hid.keyboard_layout_us import KeyboardLayoutUS
from adafruit_hid.keycode import Keycode

def configure_pin(pin):
    key_pin = digitalio.DigitalInOut(pin)
    key_pin.direction = digitalio.Direction.INPUT
    key_pin.pull = digitalio.Pull.UP
    return key_pin

key_configurations = [
    {
        "pin": configure_pin(board.GP27),
        "macro": "/time set day\n",
        "pressed": False,
    },
    {
        "pin": configure_pin(board.GP26),
        "macro": "/time set night\n",
        "pressed": False,
    },
    {
        "pin": configure_pin(board.GP28),
        "macro": "/tp 500 57 -53\n",
        "pressed": False,
    },
]

time.sleep(1)  # Sleep for a bit to avoid a race condition on some systems
keyboard = Keyboard(usb_hid.devices)
keyboard_layout = KeyboardLayoutUS(keyboard)  # We're in the US :)

while True:
    # Check each pin
    for config in key_configurations:
        if not config["pin"].value:  # Is it grounded?
            # print("Pin #%d is grounded." % i)

            if not config["pressed"]:
                keyboard_layout.write(config["macro"], 0.05)  # ...Print the string
            
            config["pressed"] = True
        else:
            # print("Pin #%d is ungrounded." % i)
            config["pressed"] = False

    time.sleep(0.05)

```

According to dmesg and LSUSB, it looks like the keyboard is recognized as a HID device, but the key strokes are not sending for some reason.  This code works fine on Windows.  I am using Linux Mint.

Any ideas on troubleshooting?

dhalbert, 2025-02-27:
Which board?
Which version of the HID library (should be latest)?
Is there anything in boot.py?

Do you see any errors in the REPL?

mcglonelevi, 2025-03-01:
> Which board?
Raspberry Pi Pico (the first version, not wireless)

> Which version of the HID library (should be latest)?
I downloaded the latest from this repo 2 days ago.

> Is there anything in boot.py?
No

> What Linux distribution and version, on what kind of computer?
```
System:
  Kernel: 6.8.0-51-generic arch: x86_64 bits: 64 compiler: gcc v: 13.3.0 clocksource: tsc
  Desktop: Cinnamon v: 6.2.9 tk: GTK v: 3.24.41 wm: Muffin v: 6.2.0 vt: 7 dm: LightDM v: 1.30.0
    Distro: Linux Mint 22 Wilma base: Ubuntu 24.04 noble
Machine:
  Type: Laptop System: System76 product: Gazelle v: gaze16-3050 serial: <superuser required>
    Chassis: type: 9 serial: <superuser required>
  Mobo: System76 model: Gazelle v: gaze16-3050 serial: <superuser required> UEFI: coreboot
    v: 2022-02-15_8c343ac date: 01/05/2022
Battery:
  ID-1: BAT0 charge: 41.6 Wh (94.1%) condition: 44.2/44.4 Wh (99.5%) volts: 16.2 min: 14.8
    model: Notebook BAT type: Li-ion serial: <filter> status: not charging cycles: 187
CPU:
  Info: 8-core model: 11th Gen Intel Core i7-11800H bits: 64 type: MT MCP smt: enabled
    arch: Tiger Lake rev: 1 cache: L1: 640 KiB L2: 10 MiB L3: 24 MiB
  Speed (MHz): avg: 1123 high: 1208 min/max: 800/4600 cores: 1: 1200 2: 1200 3: 1200 4: 1200
    5: 1068 6: 1101 7: 1200 8: 1197 9: 1208 10: 1200 11: 1029 12: 800 13: 1170 14: 800 15: 1200
    16: 1200 bogomips: 73728
  Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx
Graphics:
  Device-1: Intel TigerLake-H GT1 [UHD Graphics] vendor: CLEVO/KAPOK driver: i915 v: kernel
    arch: Gen-12.1 ports: active: eDP-1 empty: HDMI-A-1 bus-ID: 00:02.0 chip-ID: 8086:9a60
    class-ID: 0300
  Device-2: NVIDIA GA107M [GeForce RTX 3050 Mobile] vendor: CLEVO/KAPOK driver: nvidia v: 550.120
    arch: Ampere pcie: speed: 2.5 GT/s lanes: 8 ports: active: none empty: DP-1 bus-ID: 01:00.0
    chip-ID: 10de:25a2 class-ID: 0300
  Device-3: Chicony USB2.0 Camera driver: uvcvideo type: USB rev: 2.0 speed: 480 Mb/s lanes: 1
    bus-ID: 3-8:3 chip-ID: 04f2:b729 class-ID: fe01 serial: <filter>
  Display: x11 server: X.Org v: 21.1.11 with: Xwayland v: 23.2.6 driver: X:
    loaded: modesetting,nvidia unloaded: fbdev,nouveau,vesa dri: iris gpu: i915 display-ID: :0
    screens: 1
  Screen-1: 0 s-res: 1920x1080 s-dpi: 98 s-size: 499x280mm (19.65x11.02") s-diag: 572mm (22.53")
  Monitor-1: eDP-1 model: BOE Display 0x0910 res: 1920x1080 hz: 144 dpi: 142
    size: 344x194mm (13.54x7.64") diag: 395mm (15.5") modes: 1920x1080
  API: EGL v: 1.5 hw: drv: intel iris drv: nvidia platforms: device: 0 drv: nvidia device: 2
    drv: iris device: 3 drv: swrast gbm: drv: nvidia surfaceless: drv: nvidia x11: drv: iris
    inactive: wayland,device-1
  API: OpenGL v: 4.6.0 compat-v: 4.5 vendor: intel mesa v: 24.0.9-0ubuntu0.3 glx-v: 1.4
    direct-render: yes renderer: Mesa Intel UHD Graphics (TGL GT1) device-ID: 8086:9a60
Audio:
  Device-1: Intel Tiger Lake-H HD Audio vendor: CLEVO/KAPOK driver: snd_hda_intel v: kernel
    bus-ID: 00:1f.3 chip-ID: 8086:43c8 class-ID: 0403
  Device-2: NVIDIA vendor: CLEVO/KAPOK driver: snd_hda_intel v: kernel pcie: speed: 16 GT/s
    lanes: 8 bus-ID: 01:00.1 chip-ID: 10de:2291 class-ID: 0403
  API: ALSA v: k6.8.0-51-generic status: kernel-api
  Server-1: PipeWire v: 1.0.5 status: active with: 1: pipewire-pulse status: active
    2: wireplumber status: active 3: pipewire-alsa type: plugin
Network:
  Device-1: Intel Tiger Lake PCH CNVi WiFi driver: iwlwifi v: kernel bus-ID: 00:14.3
    chip-ID: 8086:43f0 class-ID: 0280
  IF: wlp0s20f3 state: up mac: <filter>
  Device-2: Realtek RTL8111/8168/8211/8411 PCI Express Gigabit Ethernet
    vendor: CLEVO/KAPOK RTL8111/8168/8411 driver: r8169 v: kernel pcie: speed: 2.5 GT/s lanes: 1
    port: 3000 bus-ID: 02:00.0 chip-ID: 10ec:8168 class-ID: 0200
  IF: enp2s0 state: down mac: <filter>
  IF-ID-1: docker0 state: down mac: <filter>
Bluetooth:
  Device-1: Intel AX201 Bluetooth driver: btusb v: 0.8 type: USB rev: 2.0 speed: 12 Mb/s lanes: 1
    bus-ID: 3-14:4 chip-ID: 8087:0026 class-ID: e001
[... 28 lines cut ...]
Sensors:
  System Temperatures: cpu: 48.0 C mobo: N/A
  Fan Speeds (rpm): cpu: 0
Repos:
  Packages: pm: dpkg pkgs: 2220
  No active apt repos in: /etc/apt/sources.list
  Active apt repos in: /etc/apt/sources.list.d/google-chrome.list
    1: deb [arch=amd64] https: //dl.google.com/linux/chrome/deb/ stable main
  Active apt repos in: /etc/apt/sources.list.d/official-package-repositories.list
    1: deb http: //packages.linuxmint.com wilma main upstream import backport
    2: deb http: //archive.ubuntu.com/ubuntu noble main restricted universe multiverse
    3: deb http: //archive.ubuntu.com/ubuntu noble-updates main restricted universe multiverse
    4: deb http: //archive.ubuntu.com/ubuntu noble-backports main restricted universe multiverse
    5: deb http: //security.ubuntu.com/ubuntu/ noble-security main restricted universe multiverse
  Active apt repos in: /etc/apt/sources.list.d/vscode.list
    1: deb [arch=amd64,arm64,armhf] https: //packages.microsoft.com/repos/code stable main
Info:
  Memory: total: 16 GiB note: est. available: 15.47 GiB used: 7.76 GiB (50.2%)
  Processes: 480 Power: uptime: 28d 3h 44m states: freeze,mem,disk suspend: deep wakeups: 2
    hibernate: platform Init: systemd v: 255 target: graphical (5) default: graphical
  Compilers: gcc: 13.3.0 Client: Cinnamon v: 6.2.9 inxi: 3.3.34
```

> Do you see any errors in the REPL?
No